LibreHardwareMonitor/LibreHardwareMonitor #1761: Mutliple sensors info missing for Ryzen 7 9700X (Temp., Power & speed) and MSI X670E Tomahawk (Fans, temp.)

StefanMacz, 2025-06-16:
Just upgraded to Ryzen 7 9700X and MSI MAG X670E Tomahawk WiFi (MS-7E12) and a number of sensors are not correctly detecting.  CPU temperature is completely missing as is CPU power and clock info as shown.

[image]

[image]

Also I note that none of the system fans or system temperatures are detected for the motherboard. MSI MAG X670E Tomahawk WiFi (MS-7E12)

LewisKaylen, 2025-06-20:
Bump2,

having a 9800X3D on Gigabyte Aorus Pro.
I am getting updates for the core clock frequency. However they seem to be off by more than 1 Ghz. Although I would appreciate a 6,7 Ghz X3D cpu.

[image]

yuxing0909, 2025-08-13:
AMD Ryzen 5 5600
AMD Ryzen 5 9600X
AMD Ryzen 7 9700X
AMD Ryzen 7 9800X3D
Moreover, the temperatures of these CPUs cannot be obtained