LibreHardwareMonitor/LibreHardwareMonitor #1765: Opt-out of Ring0

coldays, 2025-06-23:
I've been using this great library for a while on Linux, but I just started using it on windows as well.
After using it for a while, I've come to understand that if it is executed with admin privileges it installs a driver.
In my use case of this library, the information that the driver fetches is not really needed and I don't really want to be installing drivers on my users computers.

I would like to be able to configure my usage of this library so I can opt out of Computer Ring0.Open. My suggestion is either by adding a bool to Computer, extending Computer.Open with a default parameter or use the Settings that is passed to ctor of Computer.
I'm happy to create a pull request of this if the maintainers of the library agree that this is a worthwhile feature.

SolenoidMan, 2025-06-29:
2nd this, needs to be a way to opt out of the driver.

coldays, 2025-06-29:
> What kind of information do you really need ?

I'm using this for a dashboard. The information I use is CPU name, CPU total, Memory Used/Available, Upload/Download speed, GPU Video Engine, GPU Memory Used/Available, Battery charge.

As long as the opting out specifies that not all information can be fetched without the Driver, I think that is fine.
For the moment I've just forked this repo and removed the call to Ring0.Open. I've tested that I still get the information I need.

SolenoidMan, 2025-06-30:
The windows version defaults to admin only on the .exe on launch, need a fork that can be launched without admin and the driver removed. One day...

coldays, 2025-06-30:
@SolenoidMan if the program is executed without admin privileges, it will not install the driver. This can be done without any modification to the program. Here is an example using cmd which can easily be put into a bat file:
```
set __COMPAT_LAYER=RUNASINVOKER
LibreHardwareMonitor.exe
```

In this issue, my focus is on the library part because I am not in control of weather my users run my programs as admin or not.

Blacktempel, 2025-06-30:
@coldays 
You can always open a PR with that opt-out feature and see how it's liked.

@SolenoidMan 
To start without admin you can use that workaround with compat layer or locally edit [the manifest,](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitor/Resources/app.manifest) change `requestedExecutionLevel` and compile your own version.

coldays, 2025-07-09:
Pull request #1777 was just completed so I'm closing this issue. Thanks everyone!