LibreHardwareMonitor/LibreHardwareMonitor #1829: Can't tune CPU FANs with FanControl, but GCC can.

KeenBockwurst, 2025-08-22:
Hello,

i already looked into severell posts from reddit and git, but found no answer for me.
As already mentioned in the title, i cant control my cpu fans, but can with GCC.

My Hardware:

X870 AORUS ELITE WIFI7
Ryzen 7 9800X3D
RTX A4000
Corsair DDR5 DRAM 6.000 MT/s C30
Noctua NH-D15 CPU Cooler

Thanks

elmo146-ui, 2025-12-28:
same issue gigabyte aorus x870 elite wifi 7, tried different versions different .nets no matter what i just can't control any fan

Fanatic9, 2025-12-28:
> same issue gigabyte aorus x870 elite wifi 7, tried different versions different .nets no matter what i just can't control any fan

try 237v it's the one i'm working with at the moment until they figure it out , just make sure you whitelist the fancontrol folder in windows defender.