LibreHardwareMonitor/LibreHardwareMonitor #1841: Cannot read the temperatures & clocks of i9-13950HX

MitochondriaCN, 2025-09-01:
[image]

MSI Vector GP68 HX
CPU: i9-13950HX

AIDA64 can get the temperature:

[image]

phocks, 2025-09-05:
This just happened to me too. It was fine up until today, now all dashes - and can't read anything on the motherboard either

Rem0o, 2025-09-16:
See if the [driver swap](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/eb5e1a20be996d4865170b13bab97af43d97f341) fixes your issues.

closed 2025-09-27 by PhyxionNL