LibreHardwareMonitor/LibreHardwareMonitor #1867: Failed to get CPU temperature. (Intel(r) core(tm) I9 10980xe cpu 3.00ghz)

intek-jihyeon-yeom, 2025-09-17:
OS: Windows 10
CPU: Intel(r) core(tm) I9 10980xe cpu 3.00ghz
LibreHardwareMonitor: 0.9.4

UrsaMajorisDube, 2025-09-19:
I had the same problem, it was Windows Defender that blocked it. Here is what I authorized and everything works fine again.

[image]

burninc0de, 2025-09-20:
Same issue here. Adding an exclude certainly works, but it will also bypass Defender detecting that vulnerability globally, which is sub-optimal.

Should probably be patched in source.

"Open Hardware Monitor" has the same issue by the way.

leic4u, 2025-09-20:
Looking forward to the patch.

PhyxionNL, 2025-09-25:
Should work again since #1857