LibreHardwareMonitor/LibreHardwareMonitor #1872: [PawnIO] ASUS TUF X670E-PLUS WIFI shows no data + no V or MHz for 7800X3D

anirudhgupta109, 2025-09-18:
The application doesn't report any data on version [0.9.5-pre461+3c402a59ef2a87d03ae60ad714db2cfb9f3a416b](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/17801006643) for all the motherboard and some values of 7800X3d like the power, frequency and temperature. Prior to the PawnIO transition, all data and values were retrieved fine. Link of the [last working build](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions/runs/17646316801)

System specs:
TUF Gaming X670E Plus WiFi
AMD Ryzen 7800X3D

Listing the .NET i have installed:
```
$ dotnet --list-runtimes
Microsoft.NETCore.App 6.0.36 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.NETCore.App 8.0.19 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.NETCore.App 8.0.20 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.WindowsDesktop.App 6.0.36 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
Microsoft.WindowsDesktop.App 8.0.19 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
Microsoft.WindowsDesktop.App 8.0.20 [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App]
```

Report exported from the app (PawnIO version): [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/22411169/LibreHardwareMonitor.Report.txt)

Report exported from the app (WinRing0 version): [LibreHardwareMonitor.Report.txt](https://github.com/user-attachments/files/22411305/LibreHardwareMonitor.Report.txt)

Blacktempel, 2025-09-18:
@namazso is that a Windows error code ?

namazso, 2025-09-18:
Yes, see https://github.com/namazso/PawnIO.Modules/pull/26#issuecomment-3308479062

anirudhgupta109, 2025-09-18:
@namazso uninstalling Faceit AC and installing pawnio after via LHWM worked perfectly, thanks for the info, maybe a sticky in the issues is warranted or a disclaimer in case there is no workaround you can think of!

closed 2025-09-18 by anirudhgupta109