LibreHardwareMonitor/LibreHardwareMonitor #1873: Windows Defender update now suspects LibreHardwareMonitor as a threat

lord-lethris, 2025-09-18:
I think we need an update.... :)

Windows Defenders lates update now suspects LibreHardwareMonitor as a threat.  I'm sure this is a false positive.

As a side note, Windows Defender is also doing the old "your fan driver is an issue" feature again too.

UrsaMajorisDube, 2025-09-19:
I had the same problem, it was Windows Defender that blocked it. Here is what I authorized and everything works fine again.

[image]

AFract, 2025-09-21:
I am using LibreHardwareMonitor as a dependency of a personal project for workstation monitoring (temperature, drives SMART, etc), I didn't updated it since almost one year, and since a few days I have the same issue with Windows Defender complaining about it.
To fix the annoying warnings (direct update of the dependency is not an option) I set an Exclusion in Windows defender.

TeamFahQ, 2025-10-05:
I hope this is the proper place to post this
still getting trojan warnings with windows anitvirus on v0.9.4 (using the portable zip)
detected: Trojan:Win32/Vigorf.A
threat removed

AFract, 2025-11-14:
@Integrion I don't find the question dumb at all. 
The release available here was not updated since 2024, the official website https://librehardwaremonitor.com/#Download seems to point to the same outdated release, so unless there is a "nightly build" channel hidden somewhere, AFAIK the only way to test last version is to download the ZIP of for last main branch commit, and build it yourself. 
Thanks to authors, the project is highly active since september so I guess they are planning to do a new release when everything will be as good as they want :)

Fortunately it's a very easy project to build as long as you have VS2022 or 2026 installed with the proper work load, nothing more needed. This is nice compared to some open source projects requiring hours of various installation and configuration to get a working version.