LibreHardwareMonitor/LibreHardwareMonitor #1874: The latest master build doesn't show memory temperatures

jerrylzy, 2025-09-18:
The latest master build doesn't show memory temperatures. Wondering if this is a PawnIO issue? The kits are from G.Skill.

[image]

Liang-Shang, 2025-09-19:
Mine has the same issue. My MB is MSI b850m Mortar. My kits are TeamGroup T-Create Expert ones. Thx!

PhyxionNL, 2025-09-19:
cc @Blacktempel

Blacktempel, 2025-09-20:
@jerrylzy 

Thank you for the logs. Serial numbers can be removed, I don’t need to see them.

I assume “Log-1.3.1-PawnIO” is from Releases and the “Fix Test” is the version I uploaded here ?
I do see the difference but both versions seem to work properly on your system. You should be able to see RAM data in LHM.

Temperature, timings and other data are showing up normally in the logs. I would need some logging from LHM side to see if something goes wrong there. Are you able to debug LHM and check where it goes wrong on your system ?

If yes, please start here:
LibreHardwareMonitorLib/Hardware/Memory/MemoryGroup.cs

jerrylzy, 2025-09-20:
> [@jerrylzy](https://github.com/jerrylzy)
>
> Thank you for the logs. Serial numbers can be removed, I don’t need to see them.
>
> I assume “Log-1.3.1-PawnIO” is from Releases and the “Fix Test” is the [...]

Let me go through the debugger and see what went wrong

jerrylzy, 2025-09-20:
> [@jerrylzy](https://github.com/jerrylzy)
>
> Thank you for the logs. Serial numbers can be removed, I don’t need to see them.
>
> I assume “Log-1.3.1-PawnIO” is from Releases and the “Fix Test” is the [...]

I do see them in the `settings` object, but inside `DetectThermalSensors`, `detector. Accessor` is always `null`.

`settings`
[image]

`detector. Accessor`
[image]

`smbus`
[image]

`SMBusManager.RegisteredSMBuses`
[image]

Blacktempel, 2025-09-21:
@jerrylzy 
Is that a different system ?
It detects I801 so it should be an Intel system.

jerrylzy, 2025-09-21:
This might be why `pawnio_load` failed, right? It tries to load `I801`. @Blacktempel 

[image]

Blacktempel, 2025-09-21:
While explaining how things are supposed to work, I found the issue and deleted what I had written.

It seems like a try catch block was introduced in PawnIo class, which was not there when I implemented and tested on my branch.

This ultimately swallows exceptions from loading before my implementation catches and handles it. Instead of the exception it now returns a "valid" PawnIo object.

I think the current state of LHM only allows Intel to work properly, as I801 is checked first in the logic. I'll fix this when I'm back on my PC - checking all this at my phone at the moment.

jerrylzy, 2025-09-21:
@Blacktempel I already made similar changes locally. What I described [here](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1874#issuecomment-3315417835) is after I made changes locally. The issue was unrelated to LHW.

[image]

Blacktempel, 2025-09-22:
@jerrylzy 
I'll check this too later today, thanks.

Blacktempel, 2025-09-22:
@jerrylzy 
> <img alt="Image" width="1299" height="630" src="https://private-user-images.githubusercontent.com/6896568/491986735-b1066a95-edfb-4483-8d95-1b417b6bc134.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTg1MjU2NDcsIm5iZiI6MTc1ODUyNTM0NywicGF0aCI6Ii82ODk2NTY4LzQ5MTk4NjczNS1iMTA2NmE5NS1lZGZiLTQ0ODMtOGQ5NS0xYjQxN2I2YmMxMzQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjJUMDcxNTQ3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzYyNGU3NjNjOTEyOTc3NjVlNDgyM2JhMGVhYjE1NTA3MzdkNTZmZmNlOWU2ZGMxZTE3YzdjMThmNWJhNGRhZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.lTKE0l68Y1Gk_3A16S4lsdXnfgPEVI-Ic01ZjRvcOGk">

I've fixed it and added more unit tests to also include your case (different number but also a negative long).
Fix will be pushed/released later today and integrated into my PR.

Blacktempel, 2025-09-22:
I pushed changes.
Let me know if this works for you @jerrylzy