LibreHardwareMonitor/LibreHardwareMonitor #1875: AMD Ryzen 9 5900x - No clock value or Temp value being detected

thewriteway, 2025-09-19:
After updating to the latest release of LibreHardwareMonitor:

My AMD Ryzen 9 5900x doesn't show the clock and temp values anymore

For clock it says : No Value
for temp it says: 0 C

I assume this is due to the change to PawnIO?

thewriteway, 2025-09-19:
This is the Libre Hardware Monitor screenshot showing the issue with dlls built from Commit 3c402a5
[image]

thewriteway, 2025-09-19:
> For me this seems to be because Windows thinks a driver is bad
> 

That issue shouldn't apply as I'm using the latest nightly which no longer relies on Winring0 anymore.

However winring0 being removed by AV does seem to cause the exact same problem!

DanD641, 2025-09-19:
Same issue as OP with a 5900X

Powers: 0.0W
Clocks:  NaN MHz
Temp:    0.0 C

rps1408, 2025-09-20:
> For me this seems to be because Windows thinks a driver is bad
> 
> <img alt="Image" width="543" height="433" src="https://private-user-images.githubusercontent.com/7261965/491709765-37f620a0-63f1-4703-ab7a-739e59c9eaeb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgzNDYyMDUsIm5iZiI6MTc1ODM0NTkwNSwicGF0aCI6Ii83MjYxOTY1LzQ5MTcwOTc2NS0zN2Y2MjBhMC02M2YxLTQ3MDMtYWI3YS03MzllNTljOWVhZWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MjBUMDUyNTA1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDJjMjJjY2YwNzdjNmM5NjdjNjJiZDI3YzljYjJhOWZkY2QyMjllMWM0YzY5ODlmODkwZDlmYTVhMzk3MmFmMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bcZw8d5b3aN1EH66iV_Ha4TgOqqb8nOCiKXkUMZOZnY">

yup, windows is thinking winring as virus. For me it removed the file

[image]

skullboypl, 2025-09-20:
But if u will use new fork and build app Winring0 is not used anymore but still on PawnIO data is not shown like Temp CPU , cores clocks current

zapy85, 2025-09-24:
@hejrobin @FluffyRiaan download the latest nightly build https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions choose the latest one and download one of the three first links on the next page. (ie net472, net80, net90)

sidenote:
@thewriteway do you know if youll do a new build soon on the sidebardiagnostic? the og one doesnt seem to be building new ones and autostart on your fork does not seem to work?

PhyxionNL, 2025-09-24:
Everything that used to work with WinRing0 _should_ work with PawnIO as well. The only thing missing is some Gigabyte EC stuff. I don't have any of these devices to test myself unfortunately.

DanD641, 2025-09-24:
> [@hejrobin](https://github.com/hejrobin) [@FluffyRiaan](https://github.com/FluffyRiaan) habe you tried the latest nightly build https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/actions choose the latest one and download one of the three first links on the next page. (ie net472, net80, net90)
> 
> sidenote: [@thewriteway](https://github.com/thewriteway) do you know if youll do a new build soon on the sidebardiagnostic? the og one doesnt seem to be building new ones and autostart on your fork does not seem to work?

This worked for my 5900X. 
Thanks

PhyxionNL, 2025-10-15:
Check some of the PRs to see what needs to be done 😁 Closing for now as this was a driver issue.

closed 2025-10-15 by PhyxionNL